ncmreynolds/ld2410 #2: Library ld2410 dos not compile using Arduino IDE 1.8.19

MarcEngrie, 2022-11-27:
Hi, 
bought LD2410 sensor. Works using the tool 'LD2410 Tool.exe'. Now trying to use it with an ESP32.
Loaded your lib into Ardiuno IDE 1.8.19. Using the example code "setupSensor" (unchanged) and trying to compile it, generates a bunch of errors. 
Trying to use other ESP32 boards does not change anything and compiling fails with same errors. 
I attched the compile output. 
[ld2410-compile-errors.txt](https://github.com/ncmreynolds/ld2410/files/10098647/ld2410-compile-errors.txt)

Am I missing somthing or is there a setting required or ... ?

Thx in advance for helping me to solve this one

Regards
Marc Engrie

ncmreynolds, 2022-11-28:
Hi there, you seem to be using a _very_ old version of Arduino core for ESP32 (1.0.6) rather than the current version. I can compile the code fine in Arduino IDE 1.8.19 with Arduino core for ESP32 2.0.5, I used an ESP32S2 for development.

Is there a reason you are sticking with a very old version of the ESP32 core? I could probably downgrade and investigate, it looks like it's less tolerant of the argument types passed writes to the serial stream.

MarcEngrie, 2022-11-29:
Hi,
no reason why I did not upgrade beside never had the real need/reason for it. Also, I looked that I used a "old" way to embedd the ESP32 into Arduino.  
So now I replaced the link with "https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json" I got from https://randomnerdtutorials.com and it is working. So thx a lot.
Note: I am still not sure this link will keep working in the future as it refers to a 'nonn-official' Espressif link. May I ask how you make sure to use the latest SDK (and get notified of possible upgrades)?

ncmreynolds, 2022-11-29:
That looks to be the link to the stable ESP32 core, so you should get notified of updates as 'updated boards' when a new version appears. I'll close this issue now.

closed 2022-11-29 by ncmreynolds