ncmreynolds/ld2410 #5: Does not compliled at all

Kabron287, 2023-02-14:
Using library ld2410 version 0.1.1 in folder "file:///D:/Arduino/Proj/libraries/ld2410"
"D:\Arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/config" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/app_trace" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/app_update" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/asio" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/bootloader_support" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/bt" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/console" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/driver" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/efuse" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp-tls" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp32" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_adc_cal" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_event" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_http_client" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_http_server" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_https_ota" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_https_server" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_ringbuf" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_websocket_client" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/espcoredump" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/ethernet" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/expat" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/fatfs" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freemodbus" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/heap" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/idf_test" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/jsmn" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/json" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/libsodium" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/log" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/lwip" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/mbedtls" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/mdns" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/micro-ecc" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/mqtt" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/nghttp" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/nvs_flash" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/openssl" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/protobuf-c" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/protocomm" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/pthread" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/sdmmc" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/smartconfig_ack" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/soc" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/spi_flash" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/spiffs" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/tcp_transport" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/tcpip_adapter" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/ulp" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/unity" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/vfs" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/wear_levelling" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/wifi_provisioning" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/wpa_supplicant" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/xtensa-debug-module" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp-face" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp32-camera" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp-face" "-ID:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/fb_gfx" -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10809 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32_DEV\" -DARDUINO_VARIANT=\"esp32\" -DESP32 -DCORE_DEBUG_LEVEL=0 -I"D:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6\cores\esp32" -I"D:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6\variants\esp32" -I"D:\Arduino\Proj\libraries\ld2410\src" "D:\Arduino\Proj\libraries\ld2410\src\ld2410.cpp" -o "C:\Users\Vlady\AppData\Local\Temp\VMBuilds\setupSensor\esp32_esp32\Release\ld2410\ld2410.cpp.o"
ld2410.cpp: In member function bool ld2410::enter_configuration_mode_()
 
ld2410.cpp: 780:31: error: call of overloaded 'write(char)' is ambiguous
   radar_uart_->write(char(0x00))
 
Stream.h:26: In file included from
Arduino.h:147: from
ld2410.h:15: from
ld2410.cpp:15: from
Print.h:61: note  candidate  virtual size_t Print  write(uint8_t)
   virtual size_t write(uint8_t) = 0
Print.h:62: note  candidate  size_t Print  write(const char*)
   size_t write(const char *str)
 
ld2410.cpp: 782:31: error: call of overloaded 'write(char)' is ambiguous
   radar_uart_->write(char(0x00))
 
Stream.h:26: In file included from
Arduino.h:147: from
ld2410.h:15: from
ld2410.cpp:15: from
Print.h:61: note  candidate  virtual size_t Print  write(uint8_t)
   virtual size_t write(uint8_t) = 0
Print.h:62: note  candidate  size_t Print  write(const char*)
   size_t write(const char *str)
 
ld2410.cpp: 784:31: error: call of overloaded 'write(char)' is ambiguous
   radar_uart_->write(char(0x00))
 
Stream.h:26: In file included from
Arduino.h:147: from
ld2410.h:15: from
ld2410.cpp:15: from
Print.h:61: note  candidate  virtual size_t Print  write(uint8_t)
   virtual size_t write(uint8_t) = 0
Print.h:62: note  candidate  size_t Print  write(const char*)
   size_t write(const char *str)
ld2410.cpp: In member function bool ld2410::leave_configuration_mode_()
 
ld2410.cpp: 805:31: error: call of overloaded 'write(char)' is ambiguous
   radar_uart_->write(char(0x00))
 
Stream.h:26: In file included from
Arduino.h:147: from
ld2410.h:15: from
ld2410.cpp:15: from
Print.h:61: note  candidate  virtual size_t Print  write(uint8_t)
   virtual size_t write(uint8_t) = 0
Print.h:62: note  candidate  size_t Print  write(const char*)
   size_t write(const char *str)
 
ld2410.cpp: 807:31: error: call of overloaded 'write(char)' is ambiguous
[... cut ...]
   virtual size_t write(uint8_t) = 0
Print.h:62: note  candidate  size_t Print  write(const char*)
   size_t write(const char *str)
 
ld2410.cpp: 1058:32: error: call of overloaded 'write(char)' is ambiguous
   radar_uart_->write(char(0x00))
 
Stream.h:26: In file included from
Arduino.h:147: from
ld2410.h:15: from
ld2410.cpp:15: from
Print.h:61: note  candidate  virtual size_t Print  write(uint8_t)
   virtual size_t write(uint8_t) = 0
Error compiling libraries
Build failed for project 'setupSensor'

gabberhq, 2023-02-22:
got the same issue. None of the examples compile . esp8266 or esp32 give different errors. 

did you test your code ?

With esp8266 :

```

/Documents/Arduino/libraries/ld2410-main/examples/setupSensor/setupSensor.ino: In function 'void setup()':
setupSensor:19:38: error: invalid conversion from 'int' to 'SerialMode' [-fpermissive]
   19 |   Serial1.begin (256000, SERIAL_8N1, 9, 8); //UART for monitoring the radar
      |                                      ^
      |                                      |
      |                                      int
In file included from
/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/cores/esp8266/Arduino.h:303,
                 from sketch/setupSensor.ino.cpp:1:
/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.1/cores/esp8266/HardwareSerial.h:87:68: note:   initializing argument 3 of 'void HardwareSerial::begin(long unsigned int, SerialConfig, SerialMode, uint8_t)'
   87 |     void begin(unsigned long baud, SerialConfig config, SerialMode mode, uint8_t tx_pin)
      |                                                         ~~~~~~~~~~~^~~~
exit status 1
invalid conversion from 'int' to 'SerialMode' [-fpermissive]

```

ncmreynolds, 2023-02-24:
This was tested with sensors on ESP32 and AVR but not ESP8266. I'm not able to look at this for a few days but I'll try look at it on Monday. The ESP8266 has different syntax for Serial.begin so the example not working is not unexpected.

It wouldn't be much of a change to do conditional compilation in the examples that detects if it is ESP32/ESP8266.  However I didn't use this with ESP8266 as many ESP8266 development boards (eg. D1 Mini) don't make it easy to use the serial port to connect UART devices. The UART->USB chip typically has a strong pullup that prevent other devices connected to the same RX pin working. So you'd need to use Serial.swap to change the UART pins and then connect GPIO2 to TX and use Serial1 to show output. Or something along those lines. An alternative would be to use software serial for the LD2410 but I've never had great results with it and the LD2410 default baud rate is high.

ajahern95, 2023-02-24:
Arduino IDE does not like typecasting and writing 0x00 as char, byte bytes are fine.

So this doesn't work:
radar_uart_->write(char(0x00));
But writing it out as Bytes instead of chars does : 	
radar_uart_->write((byte) 0x00);

Will setup and do merge request of my variant shortly.

gabberhq, 2023-02-24:
> This was tested with sensors on ESP32 and AVR but not ESP8266. I'm not able to look at this for a few days but I'll try look at it on Monday. The ESP8266 has different syntax for Serial.begin so the [...]

Thanks for your answer. I was able to test it on esp8266 but with poor performance ( on Tesmota ). Finally the best results are with esp32.

humayunsiraj123, 2023-02-26:
Is it works with esp8266 as there's only one hardware serial and in the library serial 1-8 could be use also it doesn't works with esp8266 serial 0  .I am using D1 mini is there any way to use with any variant of 8266 
Thank you

ncmreynolds, 2023-03-01:
This should be fixed with v0.1.3 which I've done as a release.

ESP8266 support in the examples is not straightforward, so is not included. This is because the ESP8266 only has one **usable** UART and SoftwareSerial does not work with the default high baud rate of the sensor. I did try swapping the default hardware UART GPIO pins using Serial.swap() but then the ESP8266 will not boot as one of those is a strapping pin that must be low at boot but is pulled high by the LD2410.

This library should work with the ESP8266s hardware UART (the code certainly compiles) but you will need to work around the issues with strapping pins/available UART pins and the strong pullups present on most development boards like the D1 mini. As ever the ESP8266 can be fiddly to use with certain types of peripheral. I would recommend using an ESP32 or other board with multiple **usable** UARTs instead.

closed 2023-03-01 by ncmreynolds