ncmreynolds/ld2410 #12: Adding support for ESP32 S3?

Humancell, 2023-05-19:
Hello,

Thank you very much for this library!  It is great and is working well after a little troubleshooting.

I'm using a SmartBee Bee S3 (https://www.tindie.com/products/smartbee/bee-s3-esp32-s3/) and found a few issues:

1. Per some suggestions here, I had to add a 5 second delay in setup, right after the "MONITOR_SERIAL.begin(115200); //Feedback over Serial Monitor" or things didn't get output to the Serial Monitor.  This appears to be S3 related.  I found this suggestion here:  https://flashgamer.com/blog/comments/solving-platformio-issues-with-the-adafruit-feather-s3

```
  MONITOR_SERIAL.begin(115200); //Feedback over Serial Monitor
  delay(5000);
```

2. I added a new section to the pin configuration as shown below.  These were the exposed UART pins on my Bee S3:

```
    #elif CONFIG_IDF_TARGET_ESP32S3
      #define MONITOR_SERIAL Serial
      #define RADAR_SERIAL Serial1
      #define RADAR_RX_PIN 44
      #define RADAR_TX_PIN 43
```

After these changes it worked great!

ncmreynolds, 2023-05-23:
Thanks, I don't have any S3 boards so haven't tried it yet. Serial port numbering on these ESP32s with multiple hardware Serial and also the USB Serial can get a bit wordy to cover all the cases.

Humancell, 2023-05-23:
Hello ... yes, I didn't expect that you had total coverage.  It is a great library, and easy to get things working.  I wasn't super familiar with the S3 either, but it only took a little bit to get it working.

BTW, I created a little more complete sensor solution using the library, adding MQTT publishing and OTA.  https://github.com/Wovyn/esp32-ld2410-radar-sensor-mqtt-ota

Non-optimized code at this point, but your library helped this become a 3-4 hour project instead of more.  :-)

Humancell, 2023-05-23:
Also, as a side note here is a link where I found the S3 pins defined:  https://github.com/espressif/arduino-esp32/blob/3ec5f4efa1de4342aaab742008dc630091e5e035/cores/esp32/HardwareSerial.cpp#L27

closed 2023-08-26 by ncmreynolds