ncmreynolds/ld2410 #14: how do i get the individual information From each gate?

WalrusHat, 2023-08-25:


ncmreynolds, 2023-08-26:
As separately discussed, this would require adding support for engineering mode. Which I did look at briefly but never finished. Unfortunately I can't look at this for a few weeks.

ncmreynolds, 2023-09-20:
The examples in this library should work with an LD2410, have you tried them?

doc123307, 2023-09-20:
Yes, I tried to use this library. But it is written for ESP32, but I want
to use this sensor with arduino nano STM32. When copying the code, an error
occurs:
exit status 1
Compilation error: 'MONITOR_SERIAL' was not declared in this scope.

I tried to change the code to this:

#include <Arduino.h>
#include <SoftwareSerial.h> // Для создания SoftwareSerial соединения с
радаром LD2410
#include <ld2410.h>

ld2410 radar;

uint32_t lastReading = 0;
bool radarConnected = false;

#define RADAR_RX_PIN 0 // Укажите пин RX, к которому подключен радар
#define RADAR_TX_PIN 1 // Укажите пин TX, к которому подключен радар

SoftwareSerial MONITOR_SERIAL(0, 1); // Создайте SoftwareSerial соединение
для монитора
SoftwareSerial RADAR_SERIAL(RADAR_RX_PIN, RADAR_TX_PIN); // Создайте
SoftwareSerial соединение для радара

void setup() {
  MONITOR_SERIAL.begin(115200); // Начальная настройка монитора через
SoftwareSerial
  delay(500);
  MONITOR_SERIAL.println(F("\nConnect LD2410 radar TX to GPIO:"));
  MONITOR_SERIAL.println(RADAR_RX_PIN);
  MONITOR_SERIAL.println(F("Connect LD2410 radar RX to GPIO:"));
  MONITOR_SERIAL.println(RADAR_TX_PIN);
  MONITOR_SERIAL.println(F("LD2410 radar sensor initializing: "));

  // Инициализация объекта radar с использованием SoftwareSerial соединения
  if (radar.begin(RADAR_SERIAL)) {
    MONITOR_SERIAL.println(F("OK"));
    MONITOR_SERIAL.print(F("LD2410 firmware version: "));
    MONITOR_SERIAL.print(radar.firmware_major_version);
    MONITOR_SERIAL.print('.');
    MONITOR_SERIAL.print(radar.firmware_minor_version);
    MONITOR_SERIAL.print('.');
    MONITOR_SERIAL.println(radar.firmware_bugfix_version, HEX);
  } else {
    MONITOR_SERIAL.println(F("not connected"));
  }
}

void loop() {
  radar.read();
  if (radar.isConnected() && millis() - lastReading > 1000) {
    lastReading = millis();
    if (radar.presenceDetected()) {
      if (radar.stationaryTargetDetected()) {
        MONITOR_SERIAL.print(F("Stationary target: "));
        MONITOR_SERIAL.print(radar.stationaryTargetDistance());
        MONITOR_SERIAL.print(F("cm energy:"));
        MONITOR_SERIAL.print(radar.stationaryTargetEnergy());
        MONITOR_SERIAL.print(' ');
      }
      if (radar.movingTargetDetected()) {
        MONITOR_SERIAL.print(F("Moving target: "));
        MONITOR_SERIAL.print(radar.movingTargetDistance());
        MONITOR_SERIAL.print(F("cm energy:"));
        MONITOR_SERIAL.print(radar.movingTargetEnergy());
      }
      MONITOR_SERIAL.println();
    } else {
      MONITOR_SERIAL.println(F("No target"));
    }
  }
}

It was possible to upload the firmware, but in the port monitor it says:
Connect LD2410 radar TX to GPIO:
0
Connect LD2410 radar RX to GPIO:
1
LD2410 radar sensor initializing:
not connected

I also tried to change the contacts to 2.3, but nothing happened.