ncmreynolds/ld2410 #16: can't engineering mode

DuanPhan92, 2023-11-09:
When I send command `enableengineeringmode` and I alway receive `Enabling engineering mode: failed`. Can you check it?

tonhuisman, 2023-11-14:
For engineering mode to actually work you'll probably need the code from PR #3.
I've used that for a project, and it's mostly working fine, though I had to change a few bits to make it work with the way my project works, and I added a few undocumented features of the sensor.

ncmreynolds, 2023-11-15:
I'm afraid I never finished implementing engineering mode. I may come back to it at some point,. PR #3 is huge, being effectively a complete rewrite that I found impractical to meaningfully review.

tonhuisman, 2023-11-15:
As said, I am using that code, _because_ it implements the engineering mode, and it is working quite well.

If desired, I can make a new PR here with all those changes, and what I added, that should be able to merge without conflicts? Most credits should go to `skoona`, I guess 😉