ncmreynolds/ld2410 #18: Mega2560 example code

Spiridusu, 2023-12-01:
Hello. Does anyone have an example code, similar to the basicSensor example, but for the Arduino Mega 2560 using a SoftwareSerial?

tonhuisman, 2023-12-04:
Looking at the specs of that Mega 2560, it having 4 HW serial ports, and a meager 16 MHz clock-speed, why would you want to use software serial to achieve the required 256000 baud serial speed? That looks like a silly plan, as a much more stable and performant alternative is available.

Spiridusu, 2023-12-07:
I have already used 2 ports with other sensors and I want to connect 4 more sensors.

tonhuisman, 2023-12-07:
Are you installing multiple of these LD2410 sensors, or something different, that possibly (read: most likely) uses a lower baud-rate, so that could maybe use SoftwareSerial? (Just trying to find a feasible solution...)

Have you already tried to use SoftwareSerial with this LD2410 sensor? It could work, though I don't know what the load will be...