ncmreynolds/ld2410 #21: No response

bbuster2, 2024-01-07:
I am using this sensor: https://nl.aliexpress.com/item/1005006247053056.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=60454234997&aff_fcid=6006780a94294b76858c294e48c14127-1704627055686-02048-_Eumm3uD&tt=MG&aff_fsk=_Eumm3uD&aff_platform=default&sk=_Eumm3uD&aff_trace_key=6006780a94294b76858c294e48c14127-1704627055686-02048-_Eumm3uD&shareId=60454234997&businessType=ProductDetail&platform=AE&terminal_id=b880b626756d4255802651b2489a0c44&afSmartRedirect=y

I have connected 3v3 and ground to the power pins and 32 to the OT1 pin and 33 to the RX pin.
When i upload your basicSensor.ino file, i get this:

```
Connect LD2410 radar TX (OT1) to GPIO:32
Connect LD2410 radar RX to GPIO:33
LD2410 radar sensor initialising: ld2410 started

LD2410 firmware: no response
not connected
```

Also switching TX / RX pins or using different pins doesn't give me any other result.
Is my (newly bought) sensor wrong or am i doing something wrong?

ncmreynolds, 2024-01-09:
The LD2410S is a different sensor to the LD2410. I'm afraid I don't have one to test with to see if I can support it.

bbuster2, 2024-01-09:
Can i help with that? 
I have been able to receive serial data from the chip. For example:

`192 62 56 6 198 56 0 0 0 0 0 6 0 198 192 254 0 0 0 0 0 56 0 192 0 0 248 0 0 6 192 198 0 0 254 0 0 0 0 62 0 0 6 0 0 56 0 62 0 248 0 0 254 248 198 192 62 56 6 198 56 0 0 0 0 0 56 192 0 192 0 0 0 192 0 56 0 192 0 0 0 0 192 0 56 0 192 0 56 0 62 0 198 0 56 0 0 56 0 0 198 0 0 0 254 248 198 192 62 56 6 198 56 0 0 0 0 0 254 0 0 192 0 0 198 0 0 192 192 0 0 198 0 0 0 0 0 6 0 56 0 0 62 0 0 254 0 6 0 198 0 192 0 198 0 248 0 0 254 248 198 192 62 56 6 198 56 0 0 0 0 0 62 0 6 192 56 192 254 0 0 0 0 248 0 192 0 0 198 0 192 0 198 0 0 0 0 6 192 254 0 62 0 6 0 0 6 0 0 0 254 248 198 192 62 56 6 198 56 0 0 0 0 0 6 56 198 56 192 0 192 0 62 0 254 0 254 0 0 192 0 0 56 0 56 0 0 198 0 0 198 0 248 0 56 0 248 0 248 0 0 254 248 198 248 0 62 0 192 0 0 254 248 198`

If you need something i can see if i am able to get what you need

tonhuisman, 2024-01-12:
The LD2420 sends more data, as it has more (but smaller) zones, thus overflowing the receive buffer of this lib.
Also, some (newer) revisions of the LD2420 use a different communication speed from the LD2410.
You can search for LD2420 libraries to try and receive that data.

bbuster2, 2024-01-13:
I don't see any LD2420 library. Don't know if i am looking at the wrong places but i cannot find anything

tonhuisman, 2024-01-13:
Hmm, hadn't searched detailed for that. Seems that ESPHome has a working implementation, you could try that.