ncmreynolds/ld2410 #26: LD2410 to control a dry contact output relay

dritanmuco, 2024-01-26:
Hi Nick/all,

I am new here and I apologize if my question is not appropriate or clear.
Is there any way to use the output of the sensor to trigger a relay without a microcontroller?
I want to use this sensor as a motion detection sensor for an intrusion system, so I just need a clear dry output from the sensor.
I don't want to use a microcontroller because I want to keep it as small as possible and save on power consumption.

I tested one sensor by modifying the settings and was pretty nice having bluetooth, but for a security system I need the bluetooth turned off.

I hope that any of you can guide me on this.

Thank you in advance,

Humancell, 2024-01-26:
Hello,

Yes, this is not really the place for this question however I completely understand why you asked it here.  :-)

If you refer to the documentation and pinout information here:  https://www.hlktech.net/index.php?id=988

There IS an OUT pin that you could probably use for this purpose.  It goes HIGH and LOW based on if there is something detected.  I'm not sure the timing of when this occurs, how long it stays in that state, but it seems this could be used.

Now, are for driving a relay, that is another subject.  The OUT pin appears to be 3.3v, and I would guess not designed to sink a lot of current.  Many relay boards have optical isolation or gates design to drive the relay from an Arduino or ESP8266/32 type device.  These *should* work directly from the board, but no promises.  :-)

I hope this helps.

dritanmuco, 2024-01-27:
Thank you so much and sorry for posting in an inappropriate location. I am not sure where to ask this kind of question.

What I am trying to achieve is that I want to "build" a motion sensor to be installed a CAR (12V) with a NC dry contact output that I can connect to an Intruder wireless System such as DSC Neo. The problem is that I tried to use the output of the LD2410 to power an LED and it blow up the board, so that the sensor does not power anymore. I ordered another pair that I want to test, but to your point, the LD2410 Output should be triggered high in case of motion for a very short time. I am trying to find if there is a 3.3V/5V SSR relay or Mosfet that can have a NC output, but I can't find any. The dimensions should be limited and I am concerned about the power draw also.
Can I use a microcontroller (ESP32) that can sleep and wake up from the LD2410 output?
Sorry but I am newbie to ESP32 and this sensor and I am trying to understand.

Thank you again.

jenschr, 2024-03-02:
@dritanmuco As for where to ask questions: If your question is general (as in this case), I would find a suitable subforum over at https://forum.arduino.cc. I'd guess that the Device Hafking forum is a good start https://forum.arduino.cc/c/projects/device-hacking/33

If you want to use a ESP32 based microcontroller and have a question about that specific microcontroller, I would register and ask questions at https://www.esp32.com. An ESP32 can be set to "sleep" (to save power) until the sensor wakes it up using the output on the LD2410. This page could get you started on that https://randomnerdtutorials.com/esp32-deep-sleep-arduino-ide-wake-up-sources/

In general - github is not a place for support. It's just a way to maintain & distribute code.