ncmreynolds/ld2410 #27: issue with esp8266

humayunsiraj123, 2024-02-22:
firstly you have done amazing work i tested it with esp32 and it works smoothly result of stationary and moving targets, but when i tried to use with esp8266 using ESPsoftwareSerial it manage to communicate with sensor  but read wrong reasult of the stationery target but for the same position moving target is detected correctly like leas distance is 30 cm so for moving it result 30cm but from same postion stationery result around 150 to 175 cm what could be a  possible isssue  .
thank you

jenschr, 2024-03-02:
@humayunsiraj123 If you could detail a bit more about your setup, it would help solve the problem. Which devboard you are using and how you've connected the device. In general, there is no reason for the LD2410 to react any diffferently - no matter which device it's connected to.

I seriously doubt this is a library related question, but I could try to replicate the problem if you detail your setup and I have a suitable ESP8266 in one of my drawers?

humayunsiraj123, 2024-03-04:
hello , basically  i am using esp8266 d1 mini using ld2410b sensor   via
software serial  .it succesfully communicate with software serial  but the
issue when i am reading static person position  from the least distance
range of sensor i.e its returns 30 cm when i use with esp32 but when i test
it with esp8266 it results value around 150 -170 cm for static person .and
from same place it result 30 cm of motion i have the results both esp32 and
esp8266 placed on same place in front of me but returning
different results .in short while using SW serial it is measuring wrong
distacne for static target .possible its due to software serail i.e may be
software serial is missing some data in continuous detection because with
same software serial motion detection value is right .
thankyou sor for the poor english

humayunsiraj123, 2024-03-06:
[image]
following is the comparison of library with esp8266(software serail) and esp32 HW serial both the device placed on same place and detecting same person when both detects motion both shows right results but stationery person result has very big difference

RChoque, 2025-11-08:
How did you make it work (even if not precise) with the ESP-8266 ?
I can't even compile the code : I have errors like 
'pdMS_TO_TICKS' was not declared in this scope
'vTaskDelay' was not declared in this scope
'xTaskCreatePinnedToCore' was not declared in this scope

Google tells me it's because Esp-8266 is missing RTOS but I can't find a RTOS library for Arduino IDE

I also tried to compile for Arduino Uno and got the same errors.