ncmreynolds/ld2410 #32: Release 0.1.4 breaks radar.begin() return value

jenschr, 2024-10-16:
I just installed the 0.1.4 version in Arduino and radar.begin(RADAR_SERIAL) is always returning false. Setting the "waitForRadar" parameter to false will bypass the issue since it does not check the UART connection. If otherwise configured correctly, everything will work, but if code relies on the radar.begin functionality, it will fail.

Steps to reproduce:
- Open the basicSensor.ino example
- Upload to your device
- Watch the serial output to see: "LD2410 radar sensor initialising: not connected" on startup

Steps to fix:
- Use Library Manager to reinstall version 0.1.3 and all will work

Tested with ESP32C3 DevkitM 1.

Humancell, 2024-10-20:
@maxabba have you seen this?  Do you have a fix where you can provide a pull request to fix this?

maxabba, 2024-12-12:
The issue originates from the logic of the circular buffer with the ESP32 task. I will try to add a support fix for both the 'normal' routine and the task routine in the coming days.

maxabba, 2024-12-12:
I made a pull request to fix this issue