ncmreynolds/ld2410 #33: 0.1.4 Seems to break compilation with RP2040

fearthis4, 2024-10-19:
After updating to 0.1.4 I started getting the following errors during compilation:
```
Compiling library "ld2410"
"C:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\4.0.1-8ec9d6f/bin/arm-none-eabi-g++" -I "C:\\Users\\home\\AppData\\Local\\Temp\\arduino\\sketches\\29AA5119FD041ED8DE763ECF40ABB792/core" -c -Werror=return-type -Wno-psabi -DUSBD_PID=0xf00a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Generic\"" "-DUSB_PRODUCT=\"RP2040\"" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"generic\"" -DPICO_FLASH_SIZE_BYTES=2097152 "@C:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.1.1/lib/rp2040/platform_def.txt" -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -MMD "-iprefixC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.1.1/" "@C:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.1.1/lib/rp2040/platform_inc.txt" "@C:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.1.1/lib/core_inc.txt" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.1.1/include" -fno-rtti -std=gnu++17 -g -pipe -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_GENERIC_RP2040 "-DBOARD_NAME=\"GENERIC_RP2040\"" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.1.1\\cores\\rp2040" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.1.1\\variants\\generic" "-Ic:\\Users\\home\\Documents\\Arduino\\libraries\\Adafruit_NeoPixel" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.1.1\\libraries\\FreeRTOS\\src" "-Ic:\\Users\\home\\Documents\\Arduino\\libraries\\ld2410\\src" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.1.1\\libraries\\Wire\\src" "-Ic:\\Users\\home\\Documents\\Arduino\\libraries\\Adafruit_Unified_Sensor" "-Ic:\\Users\\home\\Documents\\Arduino\\libraries\\Adafruit_BME680_Library" "-Ic:\\Users\\home\\Documents\\Arduino\\libraries\\Adafruit_BusIO" "-IC:\\Users\\home\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.1.1\\libraries\\SPI\\src" "c:\\Users\\home\\Documents\\Arduino\\libraries\\ld2410\\src\\ld2410.cpp" -o "C:\\Users\\home\\AppData\\Local\\Temp\\arduino\\sketches\\29AA5119FD041ED8DE763ECF40ABB792\\libraries\\ld2410\\ld2410.cpp.o"
c:\Users\home\Documents\Arduino\libraries\ld2410\src\ld2410.cpp: In static member function 'static void ld2410::taskFunction(void*)':
c:\Users\home\Documents\Arduino\libraries\ld2410\src\ld2410.cpp:182:20: error: 'pdMS_TO_TICKS' was not declared in this scope
  182 |         vTaskDelay(pdMS_TO_TICKS(10));
      |                    ^~~~~~~~~~~~~
c:\Users\home\Documents\Arduino\libraries\ld2410\src\ld2410.cpp:182:9: error: 'vTaskDelay' was not declared in this scope
  182 |         vTaskDelay(pdMS_TO_TICKS(10));
      |         ^~~~~~~~~~
c:\Users\home\Documents\Arduino\libraries\ld2410\src\ld2410.cpp: In member function 'void ld2410::autoReadTask(uint32_t, uint32_t, uint32_t)':
c:\Users\home\Documents\Arduino\libraries\ld2410\src\ld2410.cpp:188:5: error: 'xTaskCreatePinnedToCore' was not declared in this scope
  188 |     xTaskCreatePinnedToCore(
      |     ^~~~~~~~~~~~~~~~~~~~~~~
```

fearthis4, 2024-10-19:
After reverting back to 0.1.3 the problem went away.

maxabba, 2024-12-12:
https://github.com/maxabba/ld2410/tree/Support-No-FreeRtos

Please test this branch on your RP2040 board (I don't have one for debugging) and let me know if it works. I'll submit a pull request once it's confirmed working.