ncmreynolds/ld2410 #37: Compiling on Arduino IDE fails for ESP8266 with version 0.1.4

zsalamon, 2024-12-10:
hi,

compiling for ESP8266 controllers fail after updating the library to version 0.1.4 on Arduino IDE (both v1 and v2).
part of the error message is:
"ld2410.cpp:182:20: error: 'pdMS_TO_TICKS' was not declared in this scope"

at the same time compiling with version 0.1.3 is successful.

please advise
thanks

maxabba, 2024-12-12:
https://github.com/maxabba/ld2410/tree/Support-No-FreeRtos

Please test this branch on your RP2040 board (I don't have one for debugging) and let me know if it works. I'll submit a pull request once it's confirmed working.

hcancelik, 2025-01-07:
@maxabba I have tried this with the below configs and it still throwing an error.

```platformio.ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino

; Monitor settings
monitor_speed = 115200    ; Baud rate for the Serial Monitor

lib_deps =
  https://github.com/maxabba/ld2410#Support-No-FreeRtos
```

```
In file included from .pio/libdeps/nodemcuv2/ld2410/src/ld2410.cpp:15:
.pio/libdeps/nodemcuv2/ld2410/src/ld2410.h:9:10: fatal error: freertos/FreeRTOS.h: No such file or directory
    9 | #include "freertos/FreeRTOS.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/nodemcuv2/libdec/ld2410/ld2410.cpp.o] Error 1
In file included from src/main.cpp:2:
.pio/libdeps/nodemcuv2/ld2410/src/ld2410.h:9:10: fatal error: freertos/FreeRTOS.h: No such file or directory
    9 | #include "freertos/FreeRTOS.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/nodemcuv2/src/main.cpp.o] Error 1

```

keith721, 2025-01-20:
FYI, downloaded v1.0.5 this afternoon.  Was able to successfully execute:
- radar.requestCurrentConfiguration()
- radar.requestFirmwareVersion()
Was not able to succesfully execute:
- radar.setGateSensitivityThreshold()
- radar.setMaxValues()

Thank you, developers for supporting this driver!!

zsalamon, 2026-05-11:
hi,
i installed the 0.1.4 version again using the upgrade option in Arduino IDE and tried to compile for ESP8266 but i still see this error:
"
s:\esp\libraries\ld2410\src\ld2410.cpp: In static member function 'static void ld2410::taskFunction(void*)':
s:\esp\libraries\ld2410\src\ld2410.cpp:182:20: error: 'pdMS_TO_TICKS' was not declared in this scope
  182 |         vTaskDelay(pdMS_TO_TICKS(10));
      |                    ^~~~~~~~~~~~~
s:\esp\libraries\ld2410\src\ld2410.cpp:182:9: error: 'vTaskDelay' was not declared in this scope
  182 |         vTaskDelay(pdMS_TO_TICKS(10));
      |         ^~~~~~~~~~
s:\esp\libraries\ld2410\src\ld2410.cpp: In member function 'void ld2410::autoReadTask(uint32_t, uint32_t, uint32_t)':
s:\esp\libraries\ld2410\src\ld2410.cpp:188:5: error: 'xTaskCreatePinnedToCore' was not declared in this scope
  188 |     xTaskCreatePinnedToCore(
      |     ^~~~~~~~~~~~~~~~~~~~~~~
"
version 0.1.3 compiles good.
please advise if i should use a different method for testing.
thanks

zsalamon, 2026-05-11:
> Update — v0.2.1 is now live in the Arduino Library Manager (registry
> index just refreshed). Open the IDE, search ld2410, click Update.
> Let me know if it compiles for you on the ESP8266 now.

ok, thanks for the info
let me test is tonight and give a feedback.
i have a setup i can test functionality too.